YoyPa/isw #165: /sys/kernel/debug/ec/ec0/io: No such file or directory

gezeralperen, 2021-03-10:
I can not run any command because there is no such directory as /sys/kernel/debug/ec in my Ubuntu 20.04.

gezeralperen, 2021-03-10:
After rebooting the directory appeared but this time I encountered "Operation not permitted" error for the same file "/sys/kernel/debug/ec/ec0/io"

gezeralperen, 2021-03-10:
I run the command with sudo

iFederx, 2021-04-03:
After copying all the files,
```
/etc/isw.conf
/etc/modprobe.d/isw-ec_sys.conf
/etc/modules-load.d/isw-ec_sys.conf
/usr/bin/isw
```
do this:
```
sudo modprobe -r ec_sys
sudo modprobe ec_sys write_support=1
```
then, you should be able to launch isw commands

nooballday, 2021-05-28:
@iFederx 
Shouldn't it be on documentation?

pranavraagz, 2021-07-27:
@iFederx This really should be a part of the documentation!