YoyPa/isw #166: [Support] EC Dump from MSI GF63 thin 9SC

AdemKhemiri, 2021-03-12:
EC dump
       00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F
000000 00 80 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000010 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000020 00 00 00 00 00 00 00 00 00 00 00 80 c0 06 49 0b  >..............I.<
000030 03 09 00 0d 50 0a 05 00 94 11 88 2c c2 01 e0 00  >....P......,....<
000040 00 00 64 00 b0 0f 00 00 b0 0f 6a 32 b4 0b 00 00  >..d.......j2....<
000050 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000060 00 00 00 00 00 00 00 00 24 00 37 40 49 4c 52 58  >........$.7@ILRX<
000070 64 23 23 26 2a 2d 32 37 3e 00 08 03 03 03 03 03  >d##&*-27>.......<
000080 25 00 37 3d 43 49 4f 55 63 23 23 26 2a 2d 32 37  >%.7=CIOUc##&*-27<
000090 3e 00 08 03 03 03 03 03 02 00 64 02 00 6e 00 00  >>.........d..n..<
0000a0 31 36 52 33 45 4d 53 31 2e 31 30 34 30 39 31 34  >16R3EMS1.1040914<
0000b0 32 30 32 30 31 30 3a 31 37 3a 33 32 bd 0f 40 40  >202010:17:32..@@<
0000c0 00 07 25 00 00 00 00 00 00 4c 00 00 00 e5 00 00  >..%......L......<
0000d0 00 00 40 00 7f 00 d1 00 00 00 00 00 00 00 00 00  >..@.............<
0000e0 e2 02 00 fa 10 00 00 00 00 00 00 00 00 83 00 80  >................<
0000f0 00 00 80 83 8c 00 05 00 00 00 00 00 00 00 00 00  >................<
000100

Profile dump
Value       set @ address        Fan mode
0x8c(140)   0xf4(byte244)        Advanced

-----------CPU-----------        -----------GPU-----------
Value       set @ address        Value       set @ address
0x37(55°C)  0x6a(byte106)        0x37(55°C)  0x82(byte130)
0x40(64°C)  0x6b(byte107)        0x3d(61°C)  0x83(byte131)
0x49(73°C)  0x6c(byte108)        0x43(67°C)  0x84(byte132)
0x4c(76°C)  0x6d(byte109)        0x49(73°C)  0x85(byte133)
0x52(82°C)  0x6e(byte110)        0x4f(79°C)  0x86(byte134)
0x58(88°C)  0x6f(byte111)        0x55(85°C)  0x87(byte135)

0x23(35%)   0x72(byte114)        0x23(35%)   0x8a(byte138)
0x26(38%)   0x73(byte115)        0x26(38%)   0x8b(byte139)
0x2a(42%)   0x74(byte116)        0x2a(42%)   0x8c(byte140)
0x2d(45%)   0x75(byte117)        0x2d(45%)   0x8d(byte141)
0x32(50%)   0x76(byte118)        0x32(50%)   0x8e(byte142)
0x37(55%)   0x77(byte119)        0x37(55%)   0x8f(byte143)
0x3e(62%)   0x78(byte120)        0x3e(62%)   0x90(byte144)

volnes, 2021-03-25:
dunno why u post it, but it already in latest version isw
check this on line
# GF63_8RCS GF63_9RC GF63_9RCX GF63_9SC

AdemKhemiri, 2021-03-25:
Ohh, sorry i didn't know, idk why my version doesn't have that.
Thanks tho

closed 2021-03-25 by AdemKhemiri