YoyPa/isw #181: GS65 cannot get the fans info

KoduckShih, 2021-04-19:
# Report

All function are normal except for the fan monitoring.

## isw -cp

`$ sudo isw -cp MSI_ADDRESS_DEFAULT`

~~~shell
EC dump 
       00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F
000000 00 80 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000010 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000020 00 00 00 00 00 00 00 00 00 00 00 b0 e4 06 59 0b  >..............Y.<
000030 03 01 01 05 30 0a 05 00 a0 14 60 3b 10 02 c0 00  >....0.....`;....<
000040 50 0a 33 00 e9 0f 00 00 13 08 9e 3b e4 0b f8 43  >P.3........;...C<
000050 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000060 00 00 00 00 00 00 00 00 39 00 37 40 46 4c 52 58  >........9.7@FLRX<
000070 64 00 00 00 00 46 4b 50 50 00 08 03 03 03 03 03  >d....FKPP.......<
000080 34 00 37 3d 41 47 4d 56 5e 32 00 32 3c 46 50 5a  >4.7=AGMV^2.2<FPZ<
000090 5a 00 08 03 03 03 03 05 00 00 64 02 00 6e 00 00  >Z.........d..n..<
0000a0 31 36 51 34 45 4d 53 31 2e 31 30 39 30 35 31 33  >16Q4EMS1.1090513<
0000b0 32 30 31 39 31 30 3a 34 34 3a 32 39 d9 0f 40 00  >201910:44:29..@.<
0000c0 00 06 23 00 00 9e c1 00 00 94 00 a0 00 00 14 00  >..#.............<
0000d0 00 00 00 00 70 80 00 00 05 00 00 00 00 00 00 00  >....p...........<
0000e0 e2 02 00 27 10 00 00 00 00 00 00 00 00 9a 00 bc  >...'............<
0000f0 00 00 80 00 8c 80 05 00 00 a0 32 00 00 00 00 00  >..........2.....<
000100

Profile dump 
Value       set @ address        Fan mode
0x8c(140)   0xf4(byte244)        Advanced

-----------CPU-----------        -----------GPU-----------
Value       set @ address        Value       set @ address
0x37(55°C)  0x6a(byte106)        0x37(55°C)  0x82(byte130)
0x40(64°C)  0x6b(byte107)        0x3d(61°C)  0x83(byte131)
0x46(70°C)  0x6c(byte108)        0x41(65°C)  0x84(byte132)
0x4c(76°C)  0x6d(byte109)        0x47(71°C)  0x85(byte133)
0x52(82°C)  0x6e(byte110)        0x4d(77°C)  0x86(byte134)
0x58(88°C)  0x6f(byte111)        0x56(86°C)  0x87(byte135)

0x0(0%)     0x72(byte114)        0x0(0%)     0x8a(byte138)
0x0(0%)     0x73(byte115)        0x32(50%)   0x8b(byte139)
0x0(0%)     0x74(byte116)        0x3c(60%)   0x8c(byte140)
0x46(70%)   0x75(byte117)        0x46(70%)   0x8d(byte141)
0x4b(75%)   0x76(byte118)        0x50(80%)   0x8e(byte142)
0x50(80%)   0x77(byte119)        0x5a(90%)   0x8f(byte143)
0x50(80%)   0x78(byte120)        0x5a(90%)   0x90(byte144)
~~~

## Fan Monitoring

`$ isw -r 16Q4EMS1`

~~~shell
[name@hostname ~]$ isw -r 16Q4EMS1
-----------CPU-----------        -----------GPU-----------
┌─Temp─┬─Fan Speed──────┐        ┌─Temp─┬─Fan Speed──────┐

[name@hostname ~]$ 
~~~

When i type `isw -r 16Q4EMS1`,  it only outputs the above.

How can i can the fan and thermal info?

Special-Niewbie, 2021-04-20:
Hi in meantime I will be waiting the answer in my post i jumped to your :)....

Try to get your info like that `isw -r 0` or `sudo isw -r 0` (to me it works only with administration rights as `root`)

KoduckShih, 2021-04-20:
> Hi in meantime I will be waiting the answer in my post i jumped to your :)....
> 
> Try to get your info like that `isw -r 0` or `sudo isw -r 0` (to me it works only with administration rights as `root`)

Hi Special-Niewbie ~

**isw -r 0**

In my laptop, `isw -r 0` doesn't work.
Shell outputs ...

`raise NoSectionError(section) from None configparser.NoSectionError: No section: '0'`.

**isw -h**

And the `isw -h` says ...

~~~shell
-r R      ┬ show realtime CPU+GPU temp and fan speed from EC
          └ replace R with SECTION_NAME
~~~

Seems i have a low version of `isw`?
I will update it, and then try again when I get time.
Linxu is so tiring.

Thank your comment. ~

Special-Niewbie, 2021-04-20:
I have solved everything to my laptop FINALLY........
So, if your laptop it's in the list should work without problems I have tried `sudo isw -r` without the `0` and it works the same. Because on the instruction I had read that if you add the 0 will be in loop updating the CPU and GPU info.
Anyway if your laptop it's in the list, for sure you need to check if you place in the right place all the files and write the EC with new parameter....

KoduckShih, 2021-04-21:
I updated the `isw` file in `/usr/bin`.

And `isw -r` ...

~~~shell
-----------CPU-----------        -----------GPU-----------
┌─Temp─┬─Fan Speed──────┐        ┌─Temp─┬─Fan Speed──────┐
  54°C │ 0%    0RPM     │        │ 52°C │ 0%    0RPM     │
└──────┴────────────────┘        └──────┴────────────────┘
~~~

Aha ~

Thank you for the useful info you provided !

krtk6160, 2021-05-08:
> I updated the `isw` file in `/usr/bin`.
>
> And `isw -r` ...
>
> ```shell
> -----------CPU-----------        -----------GPU-----------
> ┌─Temp─┬─Fan Speed──────┐        ┌─Temp─┬─Fan Speed──────┐
>   54°C │ [...]

Why does it show 0% and 0RPM? I have a somewhat similar issue where my GPU fan RPM is reported correctly, but my CPU fan RPM is always 0RPM.

KoduckShih, 2021-05-08:
> Why does it show 0% and 0RPM? I have a somewhat similar issue where my GPU fan RPM is reported correctly, but my CPU fan RPM is always 0RPM.

Because i set up the fan off at temperatures below 70 centigrade (both CPU GPU).
Check out if you have also made the relevant settings ~

Nate-Wilkins, 2021-06-28:
@KoduckShih Do you have steps to get this working? 
I'm running into the same issues but I don't really understand how you fixed it?