YoyPa/isw #198: EC_sys

mayank98108, 2021-06-23:
- I am getting this while using $isw -b on
 FileNotFoundError: [Errno 2] No such file or directory: '/sys/kernel/debug/ec/ec0/io'
$isw
- is working
- I have installed all files from PKGBUILD,

# modprobe ec_sys
- modprobe: FATAL: Module ec_sys not found in directory /lib/modules/5.12.11-300.fc34.x86_64

- can't load ec_sys, because it is not present in Linux kernel 5.12.11-300.fc34.x86_64

# uname -r
5.12.11-300.fc34.x86_64

- from https://github.com/torvalds/linux
- I Tried to make ec_sys.c and successfully compile ec_sys.ko, but when i try to load it

# insmod ec_sys.ko
insmod: ERROR: could not insert module ec_sys.ko: Invalid module format

- I am not able to load the kernel module manually

I am really out of options here, please help

I am using 
Fedora 34
MSI GF63 Thin 95C

kapiljhajhria, 2021-07-08:
I am having the same issue
on fedora 34 with same kernel and same system as well
Although i think maybe you made a mistake
Mine is MSI GF63 Thin 9SC

or maybe I have been reading it wrong this whole time, or these are two slightly different models. 

Were you able to fix the issue?

mayank98108, 2021-07-10:
Hi, yes it is actually MSI GF63 Thin 9SC.
No, I was not able to solve it, although I know how to solve it. It's Just I don'tṭ have that good of skill in using Linux.
One would need to compile the EC_SYS kernel module in Fedora 34 whole source code is available in Linus Torvalds Git repositories in the official Linux kernel source code directory. I was able to do that much.
After that one will need to place or configure in the installed kernel and make it part of it because currently, the problem is it is simply missing.
I need someone to help me do it and give me proper steps, I am not an advanced Linux user yet. 
In the end, I gave up and installed Arch-based Manjaro, ISW can be installed directly using yay in Manjaro. It is also running the same Linux version and it works perfectly in Manjaro.

If someone has a good understanding of Fedora, please tell us how to install a kernel module manually and use it.

kapiljhajhria, 2021-07-11:
> Hi, yes it is actually MSI GF63 Thin 9SC.
> No, I was not able to solve it, although I know how to solve it. It's Just I don'tṭ have that good of skill in using Linux.
> One would need to compile the [...]

I might move to Manjaro as well. But I really don't want to, in the end I think I might not have a choice. I was able to install the latest NVIDIA beta drivers, which did sort of showed some fan profile in GWE app, but even with that app fan speed doesn't seem to be changing.  Latest Nvidia beta driver does have fan control in its change log. I have 2 guesses, since it's beta drivers, so probably it's not working for my card properly, or it can't do what isw can do.

donteatsoclose2me, 2021-10-04:
I don't know if you ever figured this out, but the problem is that Fedora stopped including ec_sys in the normal kernel build in kernel 5.11 in Fedora 33.  To get a kernel with this module included, you'll need to install kernel-debug (which will also install kernel-debug-core, which is the package that includes. ec_sys.ko).  Once you've installed the debug kernel, you'll need to boot from it.

kapiljhajhria, 2021-10-05:
> I don't know if you ever figured this out, but the problem is that Fedora stopped including ec_sys in the normal kernel build in kernel 5.11 in Fedora 33. To get a kernel with this module included, [...]

I switched to popos.  This is the only thing that was missing in fedora except that it was perfect for me. I will give this a try when i have some free days. Although i don't know the implications of debug kernel. usually debug builds are less secure when it comes to softwares. if thats the same case with Linux kernels, then i won't be doing this.

jirislav, 2022-10-22:
Hey guys, I thought, you could maybe find this useful (especially the manual installation part, applicable to any Linux distro):
- https://github.com/YoyPa/isw/issues/265#issue-1419231642