YoyPa/isw #204: EC Dump for MSI GF75 Thin 9SC

BeardOverflow, 2021-08-04:
**MSI GF75 Thin 9SC** executing `isw -cp MSI_ADDRESS_DEFAULT`
```
EC dump 
       00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F
000000 00 80 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000010 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000020 00 00 00 00 00 00 00 00 00 00 00 80 c0 06 4a 0b  >..............J.<
000030 03 01 01 0d 50 0a 05 00 94 11 88 2c c2 01 c0 00  >....P......,....<
000040 fc 08 38 00 43 10 00 00 1b 09 b3 2d 12 0c fa 32  >..8.C......-...2<
000050 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000060 00 00 00 00 00 00 00 00 31 0a 37 40 47 4e 55 5a  >........1.7@GNUZ<
000070 64 2d 28 2d 32 37 39 41 46 00 08 03 01 05 06 03  >d-(-279AF.......<
000080 33 36 37 3d 43 49 4f 55 63 23 00 23 28 2d 32 37  >367=CIOUc#.#(-27<
000090 39 00 08 03 03 03 03 03 02 16 64 02 16 66 00 00  >9.........d..f..<
0000a0 31 37 46 32 45 4d 53 31 2e 31 30 36 30 37 31 31  >17F2EMS1.1060711<
0000b0 32 30 31 39 31 36 3a 34 35 3a 34 37 43 10 40 40  >201916:45:47C.@@<
0000c0 00 07 25 00 00 00 00 00 00 39 00 e6 00 ca 14 00  >..%......9......<
0000d0 00 00 00 00 7f 00 d1 00 00 00 00 00 00 00 00 00  >................<
0000e0 e2 02 00 f9 10 00 00 00 00 00 00 00 00 d1 00 bc  >................<
0000f0 00 00 80 83 0d 00 05 00 00 00 00 00 00 00 00 00  >................<
000100

Profile dump 
Value       set @ address        Fan mode
0xd(13)     0xf4(byte244)        Unknown

Value       set @ address        Charging below - stop @
0xbc(188)   0xef(byte239)        50% - 60%

-----------CPU-----------        -----------GPU-----------
Value       set @ address        Value       set @ address
0x37(55°C)  0x6a(byte106)        0x37(55°C)  0x82(byte130)
0x40(64°C)  0x6b(byte107)        0x3d(61°C)  0x83(byte131)
0x47(71°C)  0x6c(byte108)        0x43(67°C)  0x84(byte132)
0x4e(78°C)  0x6d(byte109)        0x49(73°C)  0x85(byte133)
0x55(85°C)  0x6e(byte110)        0x4f(79°C)  0x86(byte134)
0x5a(90°C)  0x6f(byte111)        0x55(85°C)  0x87(byte135)

0x28(40%)   0x72(byte114)        0x0(0%)     0x8a(byte138)
0x2d(45%)   0x73(byte115)        0x23(35%)   0x8b(byte139)
0x32(50%)   0x74(byte116)        0x28(40%)   0x8c(byte140)
0x37(55%)   0x75(byte117)        0x2d(45%)   0x8d(byte141)
0x39(57%)   0x76(byte118)        0x32(50%)   0x8e(byte142)
0x41(65%)   0x77(byte119)        0x37(55%)   0x8f(byte143)
0x46(70%)   0x78(byte120)        0x39(57%)   0x90(byte144)
```

Thanks a lot for this tool :+1:

marshevms, 2021-08-23:
Hi can you download rweverything on windows and create screenshot of EC? And do it for all fan mode.

BeardOverflow, 2021-08-23:
Fan basic configuration located at 0x71 position. Lowest values is 0x19 and highest value is 0x37
Fan advanced configuration is more complex to analyse. I may cover it later.
Cooler boost located at 0x98. Off 0x02, On 0x82
Switch Fn/Win key located at 0xbf. (Fn,Win) is 0x40, (Win, Fn) is 0x50
Switch Webcam located at 0x2e. Off 0x48, On 0x4a

<details>
<summary>See attached screenshots</summary>

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

</details>

marshevms, 2021-08-23:
Thank you very much

TheArcaneBrony, 2022-02-17:
Any update on this?