YoyPa/isw #223: Please add support for A5M

rathod-sahaab, 2021-11-05:
```
EC dump 
       00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F
000000 00 80 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000010 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000020 00 00 00 00 00 00 00 00 00 00 00 88 c0 06 03 0b  >................<
000030 02 05 01 0d 51 0a 05 00 f8 11 88 2c cc 01 c0 00  >....Q......,....<
000040 f8 11 26 00 be 11 d0 fe b8 06 0a 2c c9 0b fa 32  >..&........,...2<
000050 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000060 00 00 00 00 00 00 00 00 1b 00 32 3a 41 46 5a 5f  >..........2:AFZ_<
000070 64 2d 2d 3a 41 48 50 55 64 00 03 05 05 05 05 03  >d--:AHPUd.......<
000080 00 00 32 3c 46 52 5a 5d 64 00 2d 32 41 48 50 55  >..2<FRZ]d.-2AHPU<
000090 64 00 03 05 05 06 03 03 02 00 6e 02 00 66 00 00  >d.........n..f..<
0000a0 31 35 35 4c 45 4d 53 31 2e 31 30 34 30 36 30 39  >155LEMS1.1040609<
0000b0 32 30 32 31 30 39 3a 31 34 3a 32 36 be 11 00 28  >202109:14:26...(<
0000c0 00 07 25 0f 00 00 00 00 00 02 00 b0 00 ad 14 00  >..%.............<
0000d0 00 00 00 00 70 a2 00 00 1a d0 1b 80 00 00 00 00  >....p...........<
0000e0 e2 02 00 00 10 00 00 00 00 1b 80 00 00 a2 00 d0  >................<
0000f0 00 00 c1 80 0d 00 05 c1 00 00 00 00 00 00 00 00  >................<
000100

Profile dump 
Value       set @ address        Fan mode
0xd(13)     0xf4(byte244)        Unknown

-----------CPU-----------        -----------GPU-----------
Value       set @ address        Value       set @ address
0x32(50°C)  0x6a(byte106)        0x32(50°C)  0x82(byte130)
0x3a(58°C)  0x6b(byte107)        0x3c(60°C)  0x83(byte131)
0x41(65°C)  0x6c(byte108)        0x46(70°C)  0x84(byte132)
0x46(70°C)  0x6d(byte109)        0x52(82°C)  0x85(byte133)
0x5a(90°C)  0x6e(byte110)        0x5a(90°C)  0x86(byte134)
0x5f(95°C)  0x6f(byte111)        0x5d(93°C)  0x87(byte135)

0x2d(45%)   0x72(byte114)        0x2d(45%)   0x8a(byte138)
0x3a(58%)   0x73(byte115)        0x32(50%)   0x8b(byte139)
0x41(65%)   0x74(byte116)        0x41(65%)   0x8c(byte140)
0x48(72%)   0x75(byte117)        0x48(72%)   0x8d(byte141)
0x50(80%)   0x76(byte118)        0x50(80%)   0x8e(byte142)
0x55(85%)   0x77(byte119)        0x55(85%)   0x8f(byte143)
0x64(100%)  0x78(byte120)        0x64(100%)  0x90(byte144)
```

MohanVaddi, 2021-11-18:
@rathod-sahaab I'm also using A5M, please let me know if you found any solution for the fans and thermal management for linux.

arunn03, 2022-01-14:
I am too using the same model (Modern 15 A5M). Please support this model.

vlkv, 2022-02-04:
Hello,
I am thinking about buying Modern 15 A5M (AMD 5500U). Can you please tell me if the fan control problem still exist? Thx.

ng0177, 2022-04-20:
Same here. Appreciate the hard work! Please support :+1:

kuznet1, 2022-05-03:
I did some research, you can control fans if set `fan_mode` to `Advanced` (0x8C). But you can't set fans speed less than 45%, only stop if set speed to zero. It [seems](https://forum-en.msi.com/index.php?threads/latest-ec-sets-min-cpu-fan-speed-to-45-2600-rpm.319126/) like a common msi thing. Tried [155LEMS1.104](https://download.msi.com/archive/frm_exe/nb/155LEMS1.104.zip) and [155LEMS1.106](https://download.msi.com/archive/frm_exe/nb/155LEMS1.106.zip) EC firmware versions, same thing

ng0177, 2022-05-03:
True. On an MSI desktop machine w/ NVIDIA, I had to flash VBIOS. Setting the speed to zero is no issue when e.g. `Freon` `Gnome Shell Extension` is used to monitor the temperatue.