YoyPa/isw #225: Please add support for MODERN 15 A11M ( INTEL i5 )

gaionim, 2021-11-14:
`
000000 00 80 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000010 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000020 00 00 00 00 00 00 00 00 0a 05 00 00 00 04 0b 0b  >................<
000030 03 03 00 0d 01 00 50 81 f8 11 88 2c cc 01 80 00  >......P....,....<
000040 f8 11 5b 00 8c 11 10 04 d2 0f c0 32 db 0b fa 32  >..[........2...2<
000050 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000060 00 00 00 00 00 00 00 00 20 00 32 3a 41 46 5a 5f  >........ .2:AFZ_<
000070 64 2d 2d 3a 41 48 50 55 64 4b 03 05 05 05 05 03  >d--:AHPUdK......<
000080 00 00 32 3c 46 52 5a 5d 64 00 00 32 41 48 50 55  >..2<FRZ]d..2AHPU<
000090 64 46 03 05 05 06 03 03 02 00 6e 02 00 66 00 00  >dF........n..f..<
0000a0 31 35 35 32 45 4d 53 31 2e 31 31 36 30 35 32 30  >1552EMS1.1160520<
0000b0 32 30 32 31 31 30 3a 34 33 3a 30 34 00 00 00 00  >202110:43:04....<
0000c0 00 00 06 22 00 00 00 00 00 ce 00 c7 00 00 00 00  >..."............<
0000d0 00 00 c1 81 0d 00 03 80 00 01 00 00 00 00 00 00  >................<
0000e0 e2 00 00 8c 11 00 00 00 00 00 00 80 00 c2 00 00  >................<
0000f0 00 00 7f 00 00 41 37 00 41 37 00 00 00 00 00 00  >.....A7.A7......<
000100

EC dump 
[32;1m       00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F[0m

Profile dump 
[32;1mValue       set @ address        Fan mode[0m
0x0(0)      0xf4(byte244)        Unknown

[32;1mValue       set @ address        Charging below - stop @[0m
0x0(0)      0xef(byte239)        Nothing is set

-----------CPU-----------        -----------GPU-----------
[32;1mValue       set @ address        Value       set @ address[0m
0x32(50°C)  0x6a(byte106)        0x32(50°C)  0x82(byte130)
0x3a(58°C)  0x6b(byte107)        0x3c(60°C)  0x83(byte131)
0x41(65°C)  0x6c(byte108)        0x46(70°C)  0x84(byte132)
0x46(70°C)  0x6d(byte109)        0x52(82°C)  0x85(byte133)
0x5a(90°C)  0x6e(byte110)        0x5a(90°C)  0x86(byte134)
0x5f(95°C)  0x6f(byte111)        0x5d(93°C)  0x87(byte135)

0x2d(45%)   0x72(byte114)        0x0(0%)     `0x8a(byte138)`
0x3a(58%)   0x73(byte115)        0x32(50%)   0x8b(byte139)
0x41(65%)   0x74(byte116)        0x41(65%)   0x8c(byte140)
0x48(72%)   0x75(byte117)        0x48(72%)   0x8d(byte141)
0x50(80%)   0x76(byte118)        0x50(80%)   0x8e(byte142)
0x55(85%)   0x77(byte119)        0x55(85%)   0x8f(byte143)
0x64(100%)  0x78(byte120)        0x64(100%)  0x90(byte144)
`

dirompante, 2021-12-24:
Look at these two links:
[https://github.com/YoyPa/isw/issues/152](https://github.com/YoyPa/isw/issues/152)
[https://github.com/YoyPa/isw/issues/153](https://github.com/YoyPa/isw/issues/153)
Tell me if this works properly also for you (please specify your BIOS update and precise model).

gaionim, 2021-12-24:
thanks, I'll do in couple of days.
Merry Xmas.

gaionim, 2021-12-25:
Some quick test;
stop charging battery do not work :-(
Set charge to 50% 60% , parameter survive reboot, but continue charging to 100%
Try run # isw -s 0xd4 64 
but still continue charging.

PS
I'm using debian which do not has module ec_sys, so I have to boot arch linux in live cd in text mode. I can not boot in windows to test.

My model is MODERN 15 A11M-217XIT   bios E1552IMS.118

dirompante, 2021-12-26:
Thanks for your feedback. Except for `battery_charging_threshold`, does fan control work properly?
The problem with the battery threshold is that the address in EC is different from the default one used by the program.
After a couple of checks, the settings should be:
```
0xE4@0xD7 (Mobility(100-100)) = sudo isw -s 0xd7 228 
0xD0@0xD7 (Balanced(70-80)) = sudo isw -s 0xd7 208 
0xBC@0xD7 (Battery(50-60)) = sudo isw -s 0xd7 188
```
It works similarly to [https://github.com/YoyPa/isw/issues/41](https://github.com/YoyPa/isw/issues/41) but the address is different.
I would be grateful if you could give me feedback.

gaionim, 2021-12-26:
with # sudo isw -s 0xd7 208 
charging from 75 to 80 then status unknow. Capacity still 80 with charger plug in so I think battery is ok, many thanks.

For fan, I do not know how to test.
Added battery_charging_threshold = 100 to https://github.com/YoyPa/isw/issues/152 ( otherwise error )
run 
`isw -w 1552EMS1 `
`isw -s 0xd4 64 `

if I run ` isw -b on `   fan run, ` isw -b off `   fan stop , but seem the same if I reboot end run ` isw -b on `  without run `isw -w 1552EMS1 ` before.
Never tested fan before.

I'm looking for  battery_charging_threshold and I'm happy ... if no side effect ;-)

If You need some test, tell me what to do otherwise I'll close issue.

dirompante, 2021-12-26:
Many thanks for your feedback.

> Added battery_charging_threshold = 100 to #152 ( otherwise error )
Probably we are working with two different versions of this project. I have installed it from AUR. I suppose you have cloned the repo from Github. In my case, I don't need to specify that variable.
Anyway, the point is that the workaround I wrote is not the right way to fix it. You should change the line in section `[MSI_ADDRESS_DEFAULT]` of `isw.conf`:
```
battery_charging_threshold_address = 0xef
```
to
```
battery_charging_threshold_address = 0xd7
```
Similarly, instead of running `isw -s 0xd4 64` every time it should be enough to change the value of `fan_mode_address` in the same section and set it to 64 on [1552EMS1]. (honestly, I have tested this with an old version of BIOS, I don't know if is it still necessary).

> if I run isw -b on fan run, isw -b off fan stop , but seem the same if I reboot end run isw -b on without run isw -w 1552EMS1 before.
This makes sense, because you do not need to apply a profile to do that. The program writes directly on the corresponding address in [MSI_ADDRESS_DEFAULT], which fortunately is the right one for Modern 15 A11M.

I think we could integrate what we found with isw. I'm working to produce a new file "EC Rosetta" for Modern 15 A11M (similar to this [one](https://github.com/YoyPa/isw/wiki/MSI-G-laptop-EC---Rosetta).
I wrote an email to @YoyPa because I would like to modify this tool.

Finally, I would ask you to try the following in order to test the fan.
- Reboot your computer
- check if the fan is on (with your ears).
- execute the following (without executing other isw commands):
 -- `sudo isw -cp MSI_ADDRESS_DEFAULT` cut and paste the result of this
 -- `sudo isw -s 0x72 00` check if the fan turns off
 -- `sudo isw -s 0x72 80` check if the fan turns on
 -- `sudo isw -s 0x72 10` check if the fan decrease speed (the noise should decrease)
- execute the following:
 -- `sudo isw -s 0xd4 64`
 -- `sudo isw -s 0x72 00` check if the fan turns off
 -- `sudo isw -s 0x72 80` check if the fan turns on
 -- `sudo isw -s 0x72 10` check if the fan decrease speed (the noise should decrease)
After this test, you can reboot the laptop to come back to the original setting. 

I would really appreciate it if you can do this test on your computer. It should take 5 minutes.
Thanks again!

gaionim, 2021-12-26:
NB, with 
battery_charging_threshold_address = 0xd7
fan_mode_address = 0xd4

```
000000 00 80 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000010 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000020 00 00 00 00 00 00 00 00 0a 05 00 00 00 04 0b 0b  >................<
000030 02 05 00 05 00 00 50 81 f8 11 88 2c cc 01 c0 00  >......P....,....<
000040 f8 11 4d 00 84 11 f3 fd 72 0d 01 2f ba 0b fa 32  >..M.....r../...2<
000050 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000060 00 00 00 00 00 00 00 00 1a 00 32 3a 41 46 5a 5f  >..........2:AFZ_<
000070 64 00 00 3a 41 48 50 55 64 4b 03 05 05 05 05 03  >d..:AHPUdK......<
000080 00 00 32 3c 46 52 5a 5d 64 00 00 32 41 48 50 55  >..2<FRZ]d..2AHPU<
000090 64 46 03 05 05 06 03 03 02 00 6e 02 00 66 00 00  >dF........n..f..<
0000a0 31 35 35 32 45 4d 53 31 2e 31 31 36 30 35 32 30  >1552EMS1.1160520<
0000b0 32 30 32 31 31 30 3a 34 33 3a 30 34 00 00 00 00  >202110:43:04....<
0000c0 00 00 06 22 00 00 00 00 00 00 00 00 00 00 00 00  >..."............<
0000d0 00 00 c1 81 00 00 05 d0 00 01 00 00 00 00 00 00  >................<
0000e0 e2 00 00 84 11 00 00 00 00 00 00 80 00 c2 00 e4  >................<
0000f0 00 00 7f 00 00 41 37 00 41 37 00 00 00 00 00 00  >.....A7.A7......<
000100

EC dump 
       00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F

Profile dump 
Value       set @ address        Fan mode
0x0(0)      0xd4(byte212)        Unknown

Value       set @ address        Charging below - stop @
0xd0(208)   0xd7(byte215)        70% - 80%

-----------CPU-----------        -----------GPU-----------
Value       set @ address        Value       set @ address
0x32(50°C)  0x6a(byte106)        0x32(50°C)  0x82(byte130)
0x3a(58°C)  0x6b(byte107)        0x3c(60°C)  0x83(byte131)
0x41(65°C)  0x6c(byte108)        0x46(70°C)  0x84(byte132)
0x46(70°C)  0x6d(byte109)        0x52(82°C)  0x85(byte133)
0x5a(90°C)  0x6e(byte110)        0x5a(90°C)  0x86(byte134)
0x5f(95°C)  0x6f(byte111)        0x5d(93°C)  0x87(byte135)

0x0(0%)     0x72(byte114)        0x0(0%)     0x8a(byte138)
0x3a(58%)   0x73(byte115)        0x32(50%)   0x8b(byte139)
0x41(65%)   0x74(byte116)        0x41(65%)   0x8c(byte140)
0x48(72%)   0x75(byte117)        0x48(72%)   0x8d(byte141)
0x50(80%)   0x76(byte118)        0x50(80%)   0x8e(byte142)
0x55(85%)   0x77(byte119)        0x55(85%)   0x8f(byte143)
0x64(100%)  0x78(byte120)        0x64(100%)  0x90(byte144)

```

Fan off
isw -s 0x72 80  turn fan on, but low speed

isw -s 0x72 10 no change 

isw -s 0x72 00 turn fan off

 after isw -s 0xd4 64 no changes, same of above.

Run this bat in live cd

```
# da lanciare su live archlinux

# mi assicuro che sua caricato modulo
rmmod ec_sys
modprobe ec_sys write_support=1

# copio file di configurazione modificato 

cp  dirompante/isw.conf /etc/
 
 
read -p "La ventola gira ?"
 
 ./isw -cp MSI_ADDRESS_DEFAULT   | tee attuale.out
 
read -p "ora farò qualche prova"

./isw -s 0x72 00 
read -p "check if the fan turns off"

./isw -s 0x72 80 
read -p "check if the fan turns on"
./isw -s 0x72 10 
read -p "check if the fan decrease speed (the noise should decrease)"

read -p "Ora rifaccio prove dopo aver abilitato"
./isw -s 0xd4 64
./isw -s 0x72 00 
read -p "check if the fan turns off"

./isw -s 0x72 80 
read -p "check if the fan turns on"
./isw  -s 0x72 10 
read -p "check if the fan decrease speed (the noise should decrease)"

echo "finito"

```    

NB very original dump was

```
000000 00 80 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000010 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000020 00 00 00 00 00 00 00 00 0a 05 00 00 00 04 0b 0b  >................<
000030 03 03 00 0d 01 00 50 81 f8 11 88 2c cc 01 80 00  >......P....,....<
000040 f8 11 5b 00 8c 11 10 04 d2 0f c0 32 db 0b fa 32  >..[........2...2<
000050 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000060 00 00 00 00 00 00 00 00 20 00 32 3a 41 46 5a 5f  >........ .2:AFZ_<
000070 64 2d 2d 3a 41 48 50 55 64 4b 03 05 05 05 05 03  >d--:AHPUdK......<
000080 00 00 32 3c 46 52 5a 5d 64 00 00 32 41 48 50 55  >..2<FRZ]d..2AHPU<
000090 64 46 03 05 05 06 03 03 02 00 6e 02 00 66 00 00  >dF........n..f..<
0000a0 31 35 35 32 45 4d 53 31 2e 31 31 36 30 35 32 30  >1552EMS1.1160520<
0000b0 32 30 32 31 31 30 3a 34 33 3a 30 34 00 00 00 00  >202110:43:04....<
0000c0 00 00 06 22 00 00 00 00 00 ce 00 c7 00 00 00 00  >..."............<
0000d0 00 00 c1 81 0d 00 03 80 00 01 00 00 00 00 00 00  >................<
0000e0 e2 00 00 8c 11 00 00 00 00 00 00 80 00 c2 00 00  >................<
0000f0 00 00 7f 00 00 41 37 00 41 37 00 00 00 00 00 00  >.....A7.A7......<
000100

EC dump 
       00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F

Profile dump 
Value       set @ address        Fan mode
0x0(0)      0xf4(byte244)        Unknown

Value       set @ address        Charging below - stop @
0x0(0)      0xef(byte239)        Nothing is set

-----------CPU-----------        -----------GPU-----------
Value       set @ address        Value       set @ address
0x32(50°C)  0x6a(byte106)        0x32(50°C)  0x82(byte130)
0x3a(58°C)  0x6b(byte107)        0x3c(60°C)  0x83(byte131)
0x41(65°C)  0x6c(byte108)        0x46(70°C)  0x84(byte132)
0x46(70°C)  0x6d(byte109)        0x52(82°C)  0x85(byte133)
0x5a(90°C)  0x6e(byte110)        0x5a(90°C)  0x86(byte134)
0x5f(95°C)  0x6f(byte111)        0x5d(93°C)  0x87(byte135)

0x2d(45%)   0x72(byte114)        0x0(0%)     0x8a(byte138)
0x3a(58%)   0x73(byte115)        0x32(50%)   0x8b(byte139)
0x41(65%)   0x74(byte116)        0x41(65%)   0x8c(byte140)
0x48(72%)   0x75(byte117)        0x48(72%)   0x8d(byte141)
0x50(80%)   0x76(byte118)        0x50(80%)   0x8e(byte142)
0x55(85%)   0x77(byte119)        0x55(85%)   0x8f(byte143)
0x64(100%)  0x78(byte120)        0x64(100%)  0x90(byte144)
```

gaionim, 2021-12-26:
Ps  visto il mio livello di inglese forse è meglio se continuiamo in italiano ;-)

dirompante, 2021-12-26:
Ok, thanks for your reply. I have a couple of questions and we could continue privately and then write here the results.I think the

dirompante, 2021-12-30:
I post here a map of EC. The map is similar to EC Rosetta from this repository. You can find an explanation of the parameters in the second file.
I'm not completely sure about the information it contains. Use them at your own risk. In particular, there are a couple of questions I was not able to understand:
- Can the two fans be controlled separately? At the moment, GPU settings are ignored and the CPU ones are applied to both.
- What `cpu_power`,`profile_mode_1`,`profile_mode_2` really tweak?

I hope someone will integrate the program with what we have found.
[EC_reverse.pdf](https://github.com/YoyPa/isw/files/7791103/EC_reverse.pdf)
[EC_reverse_notes.pdf](https://github.com/YoyPa/isw/files/7791105/EC_reverse_notes.pdf)

Lodifice, 2022-03-07:
Hey @dirompante thanks a lot for your work! I just got a Summit B15 which is pretty similar to the Modern 15, so I decided to give your configuration a try (see #239).

While adjusting the left fan works, the right fan always keeps running and makes clearly audible noise. I suspect that left is CPU and right is GPU, correct? I can disable the right fan by writing 0 into 0x8b which according to your EC Rosetta is gpu_fan_speed_1. However, 0x8a (which should be gpu_fan_speed_0) is already 0 so I guess it's not used? I also do not understand how the controller determines that gpu_fan_speed_1 should be used, since the address for realtime_gpu_temp contains 255 (which is obviously wrong).

So my current approach for getting a usable device (that is, fans disabled when I don't do much) would be to set gpu_fan_speed_1 to 0 and also set cpu_fan_speed_0 to 0. As soon as the device heats up to about 50 degrees, the next levels of fan speed should be triggered and the fans should rapidly turn to cool the device down. Do you think this is safe to use? With fans never stopping, the device is just too noisy to be useful in a productive environment.