YoyPa/isw #236: problem when running isw -r

Shadowphoenix, 2022-02-08:
set up: 
i am running a 'ully updated Arch KDE Plasma, 
installed isw via yay
ran modprobe ec_sys as told in the readme
enabled service with systemctl enable isw@16R1EMS1.service and systemctl start isw@16R1EMS1.service without altering isw.conf

whats the issue:
upon running sudo isw -r i get the following error: 
[image]

did i do something wrong? if yes how do i fix this issue?

stef204, 2022-02-10:
what is the output of:

`# ls -l /sys/kernel/debug/ec/ec0` (as root)

Also, you might try the package `isw-git` which builds the latest `master` with the new options.

Shadowphoenix, 2022-02-12:
i just installed isw-git, after a systemctl disable, enable and reboot now it returns the expected boxes with the values. seems like the isw.1.10-package was bugged for me in some way. thanks for the recommendation - i completely forgot a git-version of this existed^^

closed 2022-02-12 by Shadowphoenix