YoyPa/isw #252: A better wiki, with clearer install instruction

Twix53791, 2022-04-28:
Hi YoyPa, first of all, thanks a lot for your program ! I could not go without the such pleasant silent it provides now ! In a sense, it's a same such a 'basic' feature (control the fans) is not implemented by default on every linux kernel... It is because it is an 'elementary' need (just a computer which not being crazily noisy all the time!) I think this program has to be the more accessible possible. Bringing to everyone the best linux experience is a very important thing, I think, to push forward the spread of linux users in the world and, I hope, one day, overthrow the domination of windows and mac on the popular computing. Anyway, an obvious thing for an experienced linux user can suck for hours a newbie (I did...), it is important to be very directive, especially to give practical examples. Isw, in fact, is not difficult to install, even I am not sure to have understand the full process. Actually, I expected the -w option to automatically create a new section in the isw.conf from the output of isw -cp MSI_ADDRESS_DEFAULT, but it seems it is not the case. I solve the problem by writing myself a new section, with the name of my motherboard, and THEN run isw -w. 

Stumbling blocks on which a common user can have trouble with:
- find the name of its motherboard and understand its meaning (is that so important? Is it in fact conventional and can it be whatever you want if you create a new one ? I think so... but I am not sure!)
- understand the typo of the commands:
     - MSI_ADDRESS_DEFAULT is itself an option, it has to be type literally
     - [SECTION], at the contrary, refers to the name of a section in isw.conf, and has to be writed without the brackets
- understand how create a new entry in isw.conf if its motherboard is not by default present

So, I propose to add the following practical example in the wiki:

### Activate isw for a MSI GS60 not supported by default
- first find the motherboard name of your MSI model on https://www.msi.com/. Just search for the name of your model.
      In my case, it is https://www.msi.com/Laptop/GS60-6QE-Ghost-Pro/support#bios. In the bios section, you find the bios version, here
      E16H7IMS.117.
- then run: `sudo isw -cp MSI_ADDRESS_DEFAULT`
- then checks if a section named 'E16H7IMS1' or something already exists in isw.conf. If its the case, just run isw -w E16...
- If its not, just create manually one, by copying another entry and editing the values. Mine is:
 
```
[16H7EMS1]
#GS60_6QE
#16H7EMS1.117
# myconfig!
address_profile = MSI_ADDRESS_DEFAULT
fan_mode = 140 
battery_charging_threshold = 100
# CPU
cpu_temp_0 = 50
cpu_temp_1 = 60
cpu_temp_2 = 68
cpu_temp_3 = 73
cpu_temp_4 = 78
cpu_temp_5 = 83
cpu_fan_speed_0 = 0
cpu_fan_speed_1 = 30
cpu_fan_speed_2 = 60
cpu_fan_speed_3 = 65
cpu_fan_speed_4 = 70
cpu_fan_speed_5 = 80
cpu_fan_speed_6 = 85```
# GPU
gpu_temp_0 = 53
gpu_temp_1 = 58
gpu_temp_2 = 63
gpu_temp_3 = 68
gpu_temp_4 = 73
gpu_temp_5 = 83
gpu_fan_speed_0 = 0
gpu_fan_speed_1 = 55
gpu_fan_speed_2 = 65
gpu_fan_speed_3 = 70
gpu_fan_speed_4 = 80
gpu_fan_speed_5 = 85
gpu_fan_speed_6 = 90
```

- Then run: `sudo isw -w 16H7EMS1`
- Then enable the isw service, according to the section name above: `systemctl enable isw@16H7EMS1.service`

thomas21120, 2022-07-03:
Thanks @Twix53791 for this extra bit of information. I almost given up on manjaro for my laptop due to lack of fan control, there is a package for ubuntu that is easy to use and works with a gui app, but I hated using ubuntu, but thanks to you and @YoyPa I got fan control working on my ge66raider-10UG.

It is also important to note that i had to run:

sudo modprobe -r ec_sys
sudo modprobe ec_sys write_support=1

to get this to work.
without i kept getting: No such file or directory: '/sys/kernel/debug/ec/ec0/io

Twix53791, 2022-08-29:
Thanks @thomas21120, actually I add these two commands to the "wiki" above, as it is true than it resolves this "No such file or directory: '/sys/kernel/debug/ec/ec0/io" error. Today, as I reinstall my arch I had to install isw again. I installed it from AUR and I got this error:
No option 'battery_charging_threshold_address' in section: 'MSI_ADDRESS_DEFAULT'.
It was the isw.conf file where the line "battery_charging_threshold_address = 0xef" was missing. I replace the file with the one from the git repo and everything works now. 

So, **I recommand to install isw from git, not from AUR**. Where to put the files? On arch, the AUR package writes the files:
/etc/modprobe.d/isw-ec_sys.conf
/etc/modules-load.d/isw-ec_sys.conf
/usr/lib/systemd/system/isw@.service
/usr/bin/isw
/etc/isw.conf

ulrickpsp, 2023-07-03:
Very useful guide. Thanks a lot.