YoyPa/isw #259: Support Katana GF66 12UG (EC ver 1583EMS1)

ferdiu, 2022-06-14:
Add support Katana GF66 12UG (EC ver 1583EMS1).

Result of command `isw -cp MSI_ADDRESS_DEFAULT`:

```
000060 00 00 00 00 00 00 00 00 3a 00 37 3e 45 4c 52 57  >........:.7>ELRW<
000070 64 2b 26 2b 30 36 3c 46 55 64 08 04 04 02 03 02  >d+&+06<FUd......<
000080 34 00 37 3d 43 48 4d 52 63 00 00 2b 30 36 3c 46  >4.7=CHMRc..+06<F<
000090 55 64 08 03 03 03 03 03 06 0f 7d 06 0a 78 37 00  >Ud........}..x7.<
0000a0 31 35 38 33 45 4d 53 31 2e 31 30 35 31 32 33 30  >1583EMS1.1051230<
0000b0 32 30 32 31 31 35 3a 32 32 3a 32 35 00 00 00 28  >202115:22:25...(<
0000c0 00 00 01 23 00 00 00 00 00 d1 00 00 00 00 00 00  >...#............<
0000d0 00 00 c1 81 0d 00 05 e4 00 05 00 00 00 00 00 00  >................<
0000e0 e2 00 00 99 11 01 00 c1 00 00 00 00 00 c0 00 00  >................<
0000f0 40 00 70 00 2f 7f 05 31 7f 05 00 00 00 01 00 00  >@.p./..1........<
000100

EC dump 
       00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F

Profile dump 
Value       set @ address        Fan mode
0x2f(47)    0xf4(byte244)        Unknown

Value       set @ address        Charging below - stop @
0x0(0)      0xef(byte239)        Nothing is set

-----------CPU-----------        -----------GPU-----------
Value       set @ address        Value       set @ address
0x37(55°C)  0x6a(byte106)        0x37(55°C)  0x82(byte130)
0x3e(62°C)  0x6b(byte107)        0x3d(61°C)  0x83(byte131)
0x45(69°C)  0x6c(byte108)        0x43(67°C)  0x84(byte132)
0x4c(76°C)  0x6d(byte109)        0x48(72°C)  0x85(byte133)
0x52(82°C)  0x6e(byte110)        0x4d(77°C)  0x86(byte134)
0x57(87°C)  0x6f(byte111)        0x52(82°C)  0x87(byte135)

0x26(38%)   0x72(byte114)        0x0(0%)     0x8a(byte138)
0x2b(43%)   0x73(byte115)        0x2b(43%)   0x8b(byte139)
0x30(48%)   0x74(byte116)        0x30(48%)   0x8c(byte140)
0x36(54%)   0x75(byte117)        0x36(54%)   0x8d(byte141)
0x3c(60%)   0x76(byte118)        0x3c(60%)   0x8e(byte142)
0x46(70%)   0x77(byte119)        0x46(70%)   0x8f(byte143)
0x55(85%)   0x78(byte120)        0x55(85%)   0x90(byte144)
```

Once this is working on my system I plan on starting the development of an `AppIndicator` to control `isw` (see: [OpenFreezeCenter](https://github.com/YoCodingMonster/OpenFreezeCenter) by [YoCodingmonster](https://github.com/YoCodingMonster) as reference for what I have in mind).

### Additional notes

I think EC updates for this laptop are released within the BIOS updates (but I may be mistaken) so I leave a list of the laptop models with share the same BIOS as listed in the _MSI BIOS Release Notes_ contained in the [BIOS update zip file](https://download.msi.com/bos_exe/nb/E1583IMS.108.zip).

MKT Name: 

- Pulse GL66 12UGK
- Pulse GL66 12UEK
- Katana GF66 12UGS
- Katana GF66 12UG
- Katana GF66 12UE
- Crosshair 15 B12UGSZ
- Crosshair 15 R6E B12UGZ
- Crosshair 15 R6E B12UEZ
- Sword 15 A12UGS
- Sword 15 A12UG
- Sword 15 A12UE
- Pulse GL66 12UGKV

levnikmyskin, 2022-06-30:
Hey! So it looks like the Katana model has a different battery threshold address. I don't know about the other addresses (since this was the only one I was interested about). I can tell you that cooler boost, for instance, work out of the box. Other things might work as well, who knows.  
  
That said, for the battery address, change the `isw.conf` file (bottom of the file, MSI_ADDRESS_DEFAULT), like:
```
...
[MSI_ADDRESS_DEFAULT]
address_profile = MSI_ADDRESS_DEFAULT
fan_mode_address = 0xf4
cooler_boost_address = 0x98
usb_backlight_address = 0xf7
battery_charging_threshold_address = 0xd7  <---- this is what you need to change (default now is 0xef)
...
```

Once you've done this and saved the file, you can simply do `isw -t 80` to set the battery threshold at 80% (laptop needs to be below 80% when you plug it in for this to work, from my experience).

ferdiu, 2022-07-02:
Hi! Thank you for this info! It is actually the main thing I needed.

Meanwhile I found other addresses and values to share:
`cooler_boost_address` is at address 0x98 (default) with values `off` 0x06 (6) and `on` 0x86 (134);
`usb_backlight_address` is at address 0xf7. Supported values are from 0x80 (off) to 0x83 (brightest).

As you said those controls works out of the box so it was really simple to find them.

Here my `diff --color=auto ~/isw/etc/isw.conf /etc/isw.conf` at this time:

```diff
1287,1288c1287,1288
< cooler_boost_off = 0
< cooler_boost_on = 128
---
> cooler_boost_off = 6
> cooler_boost_on = 134
1293,1294c1293,1294
< usb_backlight_half = 193
< usb_backlight_full = 129
---
> usb_backlight_half = 129
> usb_backlight_full = 131
1300,1301c1300,1301
< usb_backlight_address = 0xf7
< battery_charging_threshold_address = 0xef
---
> usb_backlight_address = 0xd3
> battery_charging_threshold_address = 0xd7
```

at this point just `fan_mode_address` is left to find.

Note: the `isw` script supports only 3 modes for `usb_backlight_address` but this model has 4 modes so I set `usb_backlight_half` being the least bright (0x81) and `usb_backlight_full` being the brightest (0x82) leaving the medium bright out (0x82).

levnikmyskin, 2022-07-02:
That looks nice. Can I ask you to share how you're proceeding about this? I mean, how are you finding addresses and so on.
For me, I simply "reverse engineered" by switching back and forth from windows to linux and changing battery threshold values.  
I think it'd be great to have as much info as possible by anyone who's working on this

ferdiu, 2022-07-02:
> I think it'd be great to have as much info as possible by anyone who's working on this

I totally agree with this since it took me a while to get into the process of doing this.

To my understanding there are three ways you can do this:

1. if the feature is working out of the box with the keyboard shortcut you can simply cycle through the different _states_ calling `isw -cp MSI_ADDRESS_DEFAULT` in the process to see what is changing (this is what I did to spot addresses and values for `use_backlight` and `cooler_boost`;
2. going back and forth between Windows and GNU/Linux changing parameters in Windows and dumping the EC content in Linux, as you did (this is how I tested the address you suggested for the battery threshold, sorry I couldn't test it right away since this machine is the one I am working with so I had to be extremely cautious);
3. use softwares like [RWEverything](http://rweverything.com/screenshots/) as shown [here](https://github.com/YoyPa/isw/issues/41#issuecomment-569240349) to avoid going back and forth between OSes while changing parameters from official MSI softwares such as [MSI Center](https://www.msi.com/Landing/MSI-Center).

The first method is trivial and really simple. The second and third methods are comparable but, even if it is much more painful and time consuming switching between OSes, I prefer the second because I couldn't find the source code of the suggested software anywhere and I'm not comfortable running such invasive tools without the possibility to look at the code.

Probably the third way should be "the way to go" but only if someone finds, or knows, a Windows alternative to `od` and a little bit of knowledge of the Windows system to know where to find the EC to dump it and see it changing while playing around with MSI Center. Unfortunately I don't have the time to take this journey and I'm not interested in understanding Windows systems but I hope I provided enough info to inspire somebody to do so.

ferdiu, 2022-07-03:
I have good updates: I found the tool I was looking for [ec-probe](https://github.com/hirschmann/nbfc/wiki/EC-probing-tool).

It will allow you to dump the EC while changing parameters in MSI Center and it is open-source.

Using `ec-probe` I found other addresses and values but keep in mind that I didn't test yet on Linux any of the values I'm posting here so use it really carefully and maybe test if they checks out on your Windows too before trying to apply them.

### Camera

Address: 2E
ON value (means camera is on): 0A
OFF value (means camera is off): 08

### Microphone

Address: 2C
ON value (means microphone is off): 0A
OFF value (means microphone is on): 08

### Screen Off (the one controlled with Fn+RightArrow on this model)

Address: D9
ON value (screen is off): 01
OFF value (screen is on): 05

### Speakers Volume Toggle

Address: 2D
ON value (means Volume off): 0E
OFF value (means Volume on): 0C

### Super Key Toggle (Windows key disabler)

Seems controlled by software because there are no apparent changes in the dumped EC while switching it on/off.

### Switch Fn key with Super key

Address: E8
Normal config: 00
Switched config: 10

### Switch Fn mode (the one controlled by Fn+ESC on this model)

Address: D9
OFF value (for instance: need to press Fn to change brightness): 05
ON value (for instance: no need to press Fn to change brightness): 07

### USB Power Share (allows to power USB devices while computer is off and not plugged to the wall)

Address: BF
ON value: 28
OFF value: 08

### Fan mode / User scenario

Unfortunately I'm still confused about this but I took some notes:

| User scenario | Settings | D2 | D4 | EB |
|--------------------|-------------|-----|------|--------|
| Extreme Performance | Advanced | C4 | 4D | 00 |
| Extreme Performance | Auto | C4 | 0D | 00 |
| Balanced | - | C1 | 0D | 00 |
| Silent | - | C1 | 1D | 00 |
| Super Battery | - | C2 | 0D | 0F |

Last three columns are the addresses that are changing while switching "User scenarios" in MSI Center.

It appears that D2 is somehow related with the performances (C4 = performance, C1 = Balanced, C2 = Power save maybe) and D4 with the fan mode (4D = advanced, 0D = auto, 1D = off or something like that).

Still no idea what EB does but I would bet that it is controlling some other feature (maybe disabling it) to save battery.

#### Final notes

Since in D4 the difference between 4D and 0D is exactly 64 (decimal) just like 64 was the difference between the fan modes as shown by [this if statements](https://github.com/YoyPa/isw/blob/7c88670504ecd4462b1825a55bdb0be2944dfe94/isw#L158) I feel pretty confident about being D4 the Fan mode controller indeed.

levnikmyskin, 2022-07-03:
This is amazing! Thank you so much for this comment. I think this allows us (and other people) to actually effectively develop windows-like functionality for all models in (kind of) a safe way. If you need any help or want to contact me, I'm sending you an email address with my contacts.

Good job!

levnikmyskin, 2022-07-03:
So, I have explored a bit with the tool you provided. 
I can confirm basically what you said about the addresses. However, I still need to understand whether there are things changing because they need to or just due to some other process.

So, I noticed that the (almost) whole row 90 changed between extreme and silent (basically from `90-01` to `90-0E`). I feel this is due to "noise" and not to the actual change made by MSI control center.  Can you confirm this? Everything else seems good to me.

ferdiu, 2022-07-05:
I did more testing:
there is indeed a changing value I missed in my previous post which is at the address 91 which value is 64 for Performance and Balanced modes and 53 for Silent and Super Battery modes.

I leave here two recordings annotated with what I was changing; this is the result of calling `.\ec-probe.exe monitor -i 3`:

[image]

(time advances to the right)

[image]

As you can se there are no other changes that can be directly related with changing _User Scenario_.

In the row you cited (9X) I see only the aforementioned 91 plus 9E changing with the latter changing no matter if I was changing _User Scenario_ or not. That seems to be what I would call an _output address_ since the value is changing one unit at a time. It appears to be the measurement of something, maybe.

Here it is the updated table

| User scenario | Settings | D2 | D4 | EB | 91 |
|--------------------|-------------|-----|------|--------|-----|
| Extreme Performance | Advanced | C4 | 4D | 00 | 64 |
| Extreme Performance | Auto | C4 | 0D | 00 | 64 |
| Balanced | - | C1 | 0D | 00 | 64 |
| Silent | - | C1 | 1D | 00 | 53 |
| Super Battery | - | C2 | 0D | 0F | 53 |

do not forget that all values in this table are in hex (64 is 100 decimal and 53 is 83).

ferdiu, 2022-07-05:
A small update with what I tested today under GNU/Linux:

- :heavy_check_mark:: confirmed;
- :white_check_mark:: not what I was hoping for but about right (see descriptions);
- :heavy_multiplication_x:: wrong;
- -: still not tested.

| Feature | Confirmed | Address | On | Off | Description |
|------------|---------------|-------------|------|------|----------------|
| WebCam | :heavy_check_mark: | 2E | 0A | 08 | Turn off webcam |
| Microphone | :white_check_mark: | 2C | 0A | 08 | Turn on/off microphone excluded LED (doesn't really turn off the microphone) |
| Speakers | :white_check_mark: | 2D | 0E | 0C | Turn on/off speakers excluded LED (doesn't really turn off the speakers) |
| Screen Off | :heavy_check_mark: | D9 | 01 | 05 | Turn off the screen (don't know how to test the _Off_ value since the screen turns back on as soon as you press any key) |
| Fn :left_right_arrow: Win | :heavy_check_mark: | E8 | 10 | 00 | Make `Fn` key act as `Super` and viceversa |
| Switch Fn mode | - | D9 | 07 | 05 | Toggle Fn mode (whether is needed to press Fn to access function keys or not) |
| USB Power Share | - | BF | 28 | 08 | Allows to power USB devices while computer is off and not plugged to the wall |

I will try to keep this table up-to-date.

For now I'm leaving the _Fan Mode_ out of this table since there are still too many _ifs_.

levnikmyskin, 2022-07-05:
Great job really! super useful. I'll test some stuff as well as soon as I have time

ferdiu, 2022-07-06:
Guess what: the `realtime_cpu_fan_rpm_address`es moved too! Now they are at C8-C9 instead of CC-CD:

in `/etc/isw.conf` change `realtime_cpu_fan_rpm_address`:
```
[MSI_ADDRESS_DEFAULT]
address_profile = MSI_ADDRESS_DEFAULT
[...]
realtime_cpu_fan_rpm_address = 0xc8 # it was 0xcc
[...]
```

They probably figured that it was a good idea to respect the order used before in the [schema](https://raw.githubusercontent.com/YoyPa/isw/master/wiki/msi%20ec.png) (CPU before GPU).

ferdiu, 2022-07-06:
> how did you figure this out this time?

Just turning on/off the cooler_boost to make the RPMs goes up and down and of course I was expecting them not to be so different from the GPU on (which I knew was working as shown by `isw -r`).

### Note

Actually they could have been just C9 and CD (for CPU and GPU respectively) I just assumed that the original schema was right assigning "two addresses" to the RPM addresses since the max values I registered on my laptop were 004D for the CPU and 0051 for the GPU. Those 00 could be just padding, right? I guess it doesn't really matter as long as the software which is reading the value is considering this.

ferdiu, 2022-07-17:
Hi, I have some good updates but since keeping track of all discoveries was getting harder and harder and because I wanted to check if there was some kind of pattern in the design of the EC (for instance something like a "CPU area" and a "battery area") I started drawing a schema inspired by [this](https://github.com/YoyPa/isw/blob/master/wiki/msi%20ec.png), I'm just posting here a PNG export of my spreadsheet.

Most noticeable discovery are _plugged status_, _charging status_ and _battery charge_.

Note: the text in _italic_ and wrapped inside square brackets means that the feature was either not tested or never actually seen on my computer but can be assumed since is a "well known behavior" for that particular feature (see some value for the _Shift Mode_ for instance).

[image]

You probably have to open it in another tab to see it clearly.

PS: I noticed that there are some similarities with [this](https://github.com/BeardOverflow/msi-ec/blob/main/constants.h) by [BeardOverflow](https://github.com/BeardOverflow); just adding this as reference since I used it to understand what was going on with 0xD2 (_Shift Mode_) and 0xD4 (_Fan Mode_) addresses.

levnikmyskin, 2022-07-17:
Amazing job as usual! I'm super busy these months so I don't really have time to check on all the stuff you posted...but sooner or later I'll try it out!