YoyPa/isw #269: Support for MSI Delta 15 A5EX

m-alt-f, 2023-02-23:
EC dump 
       00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F
000000 00 80 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000010 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000020 00 00 00 00 00 00 00 00 00 00 00 88 e0 06 1a 0b  >................<
000030 03 03 00 05 51 0a 05 00 a0 14 60 3b 10 02 80 00  >....Q.....`;....<
000040 18 15 23 00 09 12 f5 09 34 06 70 3e ed 0b f8 43  >..#.....4.p>...C<
000050 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000060 00 00 00 00 00 00 00 00 30 00 37 3c 46 4e 55 5a  >........0.7<FNUZ<
000070 78 00 00 2d 3c 46 4b 50 50 00 07 03 03 03 03 03  >x..-<FKPP.......<
000080 00 00 37 3c 46 50 52 57 00 00 00 37 41 46 4b 50  >..7<FPRW...7AFKP<
000090 50 00 07 03 03 03 03 03 02 00 73 00 00 66 00 00  >P.........s..f..<
0000a0 31 35 43 4b 45 4d 53 31 2e 31 30 38 30 39 30 32  >15CKEMS1.1080902<
0000b0 32 30 32 31 31 37 3a 30 39 3a 33 34 09 12 00 48  >202117:09:34...H<
0000c0 00 02 23 27 d0 9e 40 00 00 6e 00 00 00 00 14 00  >..#'..@..n......<
0000d0 00 00 00 00 00 a0 00 00 27 00 31 80 00 00 00 00  >........'.1.....<
0000e0 e2 02 00 01 10 00 00 00 00 00 27 80 00 84 00 80  >..........'.....<
0000f0 00 00 c1 83 1d 00 05 00 00 00 00 00 00 00 00 00  >................<
000100

Profile dump 
Value       set @ address        Fan mode
0x1d(29)    0xf4(byte244)        Unknown

-----------CPU-----------        -----------GPU-----------
Value       set @ address        Value       set @ address
0x37(55°C)  0x6a(byte106)        0x37(55°C)  0x82(byte130)
0x3c(60°C)  0x6b(byte107)        0x3c(60°C)  0x83(byte131)
0x46(70°C)  0x6c(byte108)        0x46(70°C)  0x84(byte132)
0x4e(78°C)  0x6d(byte109)        0x50(80°C)  0x85(byte133)
0x55(85°C)  0x6e(byte110)        0x52(82°C)  0x86(byte134)
0x5a(90°C)  0x6f(byte111)        0x57(87°C)  0x87(byte135)

0x0(0%)     0x72(byte114)        0x0(0%)     0x8a(byte138)
0x2d(45%)   0x73(byte115)        0x37(55%)   0x8b(byte139)
0x3c(60%)   0x74(byte116)        0x41(65%)   0x8c(byte140)
0x46(70%)   0x75(byte117)        0x46(70%)   0x8d(byte141)
0x4b(75%)   0x76(byte118)        0x4b(75%)   0x8e(byte142)
0x50(80%)   0x77(byte119)        0x50(80%)   0x8f(byte143)
0x50(80%)   0x78(byte120)        0x50(80%)   0x90(byte144)

m-alt-f, 2023-02-23:
Hi @YoyPa - I have no idea how hard it would be to add support for this particular laptop.
I'm just putting this here in case it's of some use!

jpstierlin, 2023-02-27:
It's easy once you know what to do with these values, just copy another section in your /etc/isw.conf file, name it 15CKEMS1 from you laptop's dump, and customize it with the given values, then activate the systemd service with sudo systemctl enable isw@SECTION.service.
Too bad this project seems abandoned.

TeaJamFA, 2023-09-04:
I have the exact same laptop with a nearly identical EC dump using the same addresses as well. All of the sensors seem to be working, and Cooler Booster works as well. The GPU fan adjusts properly however the CPU fan does not. It's fan curve seems to be locked based on the actual temperature with no way to control it. I have also tried to control it in the bios and the option didn't even work either. It seems to be the same deal on Windows with the official MSI Center app as well. @m-alt-f have you also had these issues and @jpstierlin do you know what may be causing them?