YoyPa/isw #279: MSI GS66 Stealth 10UG

Sebron, 2024-06-01:
outout from 'sudo isw -cp MSI_ADDRESS_DEFAULT'

EC dump
00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F
000000 00 80 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000010 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000020 00 00 00 00 00 00 00 00 00 00 00 80 c8 06 0b 4b  >...............K<
000030 03 09 01 0d 50 0a 05 00 6a 18 60 3b 71 02 e0 00  >....P...j.`;q...<
000040 00 00 60 00 d6 15 00 00 c7 14 47 41 21 0c 00 00  >..`.......GA!...<
000050 03 c0 00 02 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000060 00 00 00 00 00 00 00 00 2e 00 37 3c 41 46 5a 5f  >..........7<AFZ_<
000070 64 2d 2d 32 3c 41 50 55 64 00 08 07 05 05 05 03  >d--2<APUd.......<
000080 2c 00 32 3c 46 52 5a 5d 64 00 00 2d 3c 46 50 55  >,.2<FRZ]d..-<FPU<
000090 64 00 03 05 05 06 03 03 82 00 7d 02 00 64 00 00  >d.........}..d..<
0000a0 31 36 56 33 45 4d 53 31 2e 31 30 36 30 37 32 31  >16V3EMS1.1060721<
0000b0 32 30 32 31 31 33 3a 35 36 3a 31 32 d6 15 00 28  >202113:56:12...(<
0000c0 00 07 22 00 00 9e c0 00 00 bf 00 37 00 37 00 00  >.."........7.7..<
0000d0 00 00 00 00 70 00 00 00 2c 00 00 80 00 18 06 01  >....p...,.......<
0000e0 e2 02 00 59 10 00 00 00 00 00 00 00 00 9a 00 80  >...Y............<
0000f0 00 00 80 83 0d 00 05 00 00 3a 00 00 00 00 00 00  >.........:......<
000100

Profile dump
Value       set @ address        Fan mode
0xd(13)     0xf4(byte244)        Unknown

-----------CPU-----------        -----------GPU-----------
Value       set @ address        Value       set @ address
0x37(55°C)  0x6a(byte106)        0x32(50°C)  0x82(byte130)
0x3c(60°C)  0x6b(byte107)        0x3c(60°C)  0x83(byte131)
0x41(65°C)  0x6c(byte108)        0x46(70°C)  0x84(byte132)
0x46(70°C)  0x6d(byte109)        0x52(82°C)  0x85(byte133)
0x5a(90°C)  0x6e(byte110)        0x5a(90°C)  0x86(byte134)
0x5f(95°C)  0x6f(byte111)        0x5d(93°C)  0x87(byte135)

0x2d(45%)   0x72(byte114)        0x0(0%)     0x8a(byte138)
0x32(50%)   0x73(byte115)        0x2d(45%)   0x8b(byte139)
0x3c(60%)   0x74(byte116)        0x3c(60%)   0x8c(byte140)
0x41(65%)   0x75(byte117)        0x46(70%)   0x8d(byte141)
0x50(80%)   0x76(byte118)        0x50(80%)   0x8e(byte142)
0x55(85%)   0x77(byte119)        0x55(85%)   0x8f(byte143)
0x64(100%)  0x78(byte120)        0x64(100%)  0x90(byte144)

Kayo4life, 2024-08-03:
Did you fix the issue, and does your laptop work with this? I have the MSI GS66 Stealth 10UG-219 US

Kayo4life, 2024-09-07:
Here is an EC dump before changing anything
`sudo isw -c
[sudo] password for kayo: 

EC dump 
       00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F
000000 00 80 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000010 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000020 00 00 00 00 00 00 00 00 00 00 00 80 c8 06 53 4b  >..............SK<
000030 03 01 00 0d 50 0a 05 00 6a 18 60 3b 71 02 c0 00  >....P...j.`;q...<
000040 35 0c 47 00 a4 10 00 00 b0 0b 35 3c a4 0b f8 43  >5.G.......5<...C<
000050 00 00 00 01 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000060 00 00 00 00 00 00 00 00 2b 00 37 3c 41 46 5a 5f  >........+.7<AFZ_<
000070 64 2d 2d 32 3c 41 50 55 64 00 08 07 05 05 05 03  >d--2<APUd.......<
000080 2b 00 32 3c 46 52 5a 5d 64 00 00 2d 3c 46 50 55  >+.2<FRZ]d..-<FPU<
000090 64 00 03 05 05 06 03 03 02 00 7d 02 00 64 00 00  >d.........}..d..<
0000a0 31 36 56 33 45 4d 53 31 2e 31 30 32 31 32 31 30  >16V3EMS1.1021210<
0000b0 32 30 32 30 31 33 3a 32 35 3a 32 35 a4 10 00 28  >202013:25:25...(<
0000c0 00 07 24 00 00 9e c0 00 00 85 00 00 00 b2 14 00  >..$.............<
0000d0 00 00 00 00 70 00 00 00 2b 00 00 80 00 18 09 07  >....p...+.......<
0000e0 e2 02 00 49 10 00 00 00 00 00 00 00 00 9a 00 d0  >...I............<
0000f0 00 00 80 83 0d 00 05 00 00 00 00 00 00 00 00 00  >................<
000100
`

Here is an EC dump after enabling cooler booster with the FN+F8 shortcut (not with the ISW command)
` sudo isw -c
[sudo] password for kayo: 

EC dump 
       00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F
000000 00 80 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000010 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000020 00 00 00 00 00 00 00 00 00 00 00 80 c8 06 53 4b  >..............SK<
000030 03 01 01 0d 50 0a 05 00 6a 18 60 3b 71 02 c0 00  >....P...j.`;q...<
000040 35 0c 47 00 a4 10 00 00 b2 0b 35 3c ae 0b f8 43  >5.G.......5<...C<
000050 00 00 00 01 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000060 00 00 00 00 00 00 00 00 28 00 37 3c 41 46 5a 5f  >........(.7<AFZ_<
000070 64 2d 2d 32 3c 41 50 55 64 00 08 07 05 05 05 03  >d--2<APUd.......<
000080 26 00 32 3c 46 52 5a 5d 64 00 00 2d 3c 46 50 55  >&.2<FRZ]d..-<FPU<
000090 64 00 03 05 05 06 03 03 82 00 7d 02 00 64 00 00  >d.........}..d..<
0000a0 31 36 56 33 45 4d 53 31 2e 31 30 32 31 32 31 30  >16V3EMS1.1021210<
0000b0 32 30 32 30 31 33 3a 32 35 3a 32 35 a4 10 00 28  >202013:25:25...(<
0000c0 00 07 24 00 00 9e c0 00 00 9d 00 3f 00 42 14 00  >..$........?.B..<
0000d0 00 00 00 00 70 00 00 00 26 00 00 80 00 18 09 07  >....p...&.......<
0000e0 e2 02 00 49 10 00 00 00 00 00 00 00 00 9a 00 d0  >...I............<
0000f0 00 00 80 83 0d 00 05 00 00 46 00 00 00 00 00 00  >.........F......<
000100
`
The last non blank value alternates between G, F, and E, but it's usually F.